MrYsLab/PyMata #8: exceptions when runnini on arduino nano clone

bschwand, 2015-02-06:
Hello,

this is with Pymata pulled from git as of a78612b94387c8ca3dad42b7f92a71819efa4f6e .
I am using an arduino nano 3.0 clone, as sold by Baite electronics ( http://goo.gl/awSni2 , although the one I have is the previous version based on FT232R chip instead of CH340). Installed latest firmata 2.4.0 from the firmata releases. I tested Firmata with snap4Arduino, and it seems to work except for the PWM and servo not functioning properly, however this served to confirm firmata is up and running on the arduino.

Here is the error:

```
~/PyMata/examples$ ./pymata_blink.py 

Python Version 2.7.3 (default, Sep 26 2013, 20:08:41) 
[GCC 4.6.3]

PyMata version 2.02    Copyright(C) 2013-15 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyUSB0 
Please wait while Arduino is being detected. This can take up to 30 seconds ...
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/PyMata/pymata_command_handler.py", line 825, in run
    method(command_data)
  File "/usr/local/lib/python2.7/dist-packages/PyMata/pymata_command_handler.py", line 385, in analog_message
    self.analog_response_table[data[self.RESPONSE_TABLE_MODE]][self.RESPONSE_TABLE_PIN_DATA_VALUE]
IndexError: list index out of range

^CProgram Aborted Before PyMata Instantiated

 ```

bschwand, 2015-02-06:
a small update:

I tested also on a different arduino nano clone, with a CH341 based serial - usb adapter like this : 
http://goo.gl/15xFyF  
http://goo.gl/fLhU9O

exact same result, so I am thinking something is triggered in Pymata, that maybe not directly related to the arduino type ?

MrYsLab, 2015-02-06:
@bshwand This is the same behvavior being seen in [issue 7](https://github.com/MrYsLab/PyMata/issues/7). I am waiting to receive the pololu board to see if I can reproduce it. 

Are you using StandardFirmata or FirmataPlus on the Arduino? If you are using StandardFirmata, which version are you using? If you are not sure, which version of the Arduino software are you using to load the Nano?

One thing I noticed, is that you are using an older version of Python. I am running Ubuntu 14.10 that uses Python 2.7.8 and GCC 4.9. I do not know if this is the cause of the problem yet, but it might be.

Lastly what operating system and version of the operating system are you using?

bschwand, 2015-02-06:
I am using Standard firmata, I updated to the version available here
https://github.com/firmata/arduino/releases/tag/v2.4.0
I built with the arduino 1.0.6 version (replaced the libraries that came with arduino). I also tried using the firmata provided with arduino 1.0.6, the result is exactly the same.
I am using ubuntu 12.04 (I need that version to run linuxcnc simulator), python 2.7.3
I will see if it is possible to update python. Let me try in a VM with ubuntu14.
Where are you located ? I could send you one of those arduino boards if you want.

bschwand, 2015-02-06:
I can trz FirmataPlus. Where can I download it ?

MrYsLab, 2015-02-06:
Thanks for the offer. The pololu board is scheduled for delivery today (a week late), so I will try that to see if I can reproduce the problem.  I downloaded12.04 and will install that in a VM as well.

The Arduino software should be good. No need to try FirmataPlus, but it does give you some additional funtionality - It provides support for:
    Tone
    Stepper
    NewPing (HC-SR04 support)
    Encoder - Note that the encoder works for an Arduino UNO and does not function
              with a Leonardo. It may not work for others boards as well.

If you want to use it, it is included with PyMata in the ArduinoSketch directory. Install instructions are included on page 12 of the PyMata Install instructions included int the PyMata Documentation directory.

By the way, which kernel are you currently using? You can find that out by issuing:
uname -r
in a terminal Window.

MrYsLab, 2015-02-06:
@bschwand I installed Ubuntu 12.04 without any updates, ran with python 2.7.3 and using a Leonardo clone, did not encounter the problem, so most likely the vintage of python, gcc or kernel is not the source of the problem. Please don't waste your time upgrading Python.

Hopefully when the pololu board appears, I will be able to reproduce the problem.

bschwand, 2015-02-06:
to answer your previous question:

bruno@vostro1500:~/PyMata/ArduinoSketch$ uname -r
3.11.0-15-generic

bschwand, 2015-02-06:
Success ! (but there is nonetheless a bug somewhere  :-)) It looks like some interaction between the hardware and the specific firmata is causing the issue. I flashed FirmataPlus to the arduino and things started working.

Maybe the bug is actually in Firmata ? Or, since you both develop FirmataPlus and PyMata, some assumption is common to both but not actually true, and revealed by the Firmata firmware ?

```
bruno@vostro1500:~/PyMata/examples$ ./pymata_blink.py 

Python Version 2.7.3 (default, Sep 26 2013, 20:08:41) 
[GCC 4.6.3]

PyMata version 2.02    Copyright(C) 2013-15 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyUSB0 
Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 22
Total Number of Analog Pins Detected = 8
Blinking LED on pin 13
1
2
3
4
5
6
7
8
9
10
PyMata close(): Calling sys.exit(0): Hope to see you soon!
```

MrYsLab, 2015-02-06:
Great information. I will need to test Firmata 2.4 which is pretty recent. FirmataPlus is just an extension of Firmata 2.3, so I am suspecting one of the "bug" fixes of 2.4 is causing the issue. Hopefully this will surface in my environment and if it is a bug in Firmata I will report, or if it can be fixed in PyMata I will do so.
Thanks again.

MrYsLab, 2015-02-07:
Firmata 2.4 is indeed causing the problem. I have entered an issue for Firmata [here](https://github.com/firmata/arduino/issues/185).

bschwand, 2015-02-07:
Excellent !
Glad this was figured out. 
And now that I discovered FirmataPlus, I will stick with it anyway, since it has great additional features. Thanks to PyMata I could verify everything works as expected ! Now I just need to find a fix for [this] (https://github.com/edutec/Snap4Arduino/issues/50) issue which I now have no doubt is in Snap4Arduino

MrYsLab, 2015-02-07:
There is a fix in the works for Firmata 2.4 that should solve the problem. BTW, [s2a_fm](https://github.com/MrYsLab/s2a_fm) and [Xi](https://github.com/MrYsLab/xi) both play well with Snap!. Xi allows multiple board types (Arduino, RaspberryPi , BeagleBone Black) to be supported simultaneously using either Snap! or Scratch.

bschwand, 2015-02-07:
thanks, I will check them out.
What I liked better about snap4arduino, is that it is a self-contained
application that can be installed to a computer, no need of internet or
server, etc.
if SNAP can be repackaged that way, that could be a solution for me.

Thanks

Bruno

MrYsLab, 2015-02-07:
The Firmata development team have agreed that this is a bug in Firmata 2.4 and will be providing a fix.

closed 2015-02-07 by MrYsLab