MrYsLab/PyMata #11: Arduino Yun OpenWRT Board Auto Discovery Failed

zawilliams, 2015-03-26:
So I am having a bit of an issue firing up the pymata_blink.py example. I know that the port is good to go because I was using pyFirmata and can communicate fine to the MCU from Python.

Here's what's coming back after running the script:

```
PyMata version 2.05   Copyright(C) 2013-15 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyATH0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board Auto Discovery Failed!, Shutting Down
Traceback (most recent call last):
  File "pymata.py", line 22, in <module>
    firmata.set_pin_mode(BOARD_LED, firmata.OUTPUT, firmata.DIGITAL)
  File "/usr/lib/python2.7/site-packages/PyMata/pymata.py", line 733, in set_pin_mode
    self._command_handler.send_command(command)
  File "/usr/lib/python2.7/site-packages/PyMata/pymata_command_handler.py", line 633, in send_command
    self.pymata.transport.write(data)
  File "/usr/lib/python2.7/site-packages/PyMata/pymata_serial.py", line 107, in write
    self.arduino.write(data)
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 490, in write
    if not self._isOpen: raise portNotOpenError
serial.serialutil.SerialException: Attempting to use a port that is not open
```

Anyone have any ideas?

zawilliams, 2015-03-26:
Ok so I found out the issue was the baud rate my Firmata sketch was running at. Upon checking checking pymata_serial.py I found the baud rate was 57600 and my Firmata sketch had 115200 set as the baud rate. Changed the sketch to have a baud rate of 57600 and it works now.

Once the script finished this happened though. Should I worry about this?

```
Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 30
Total Number of Analog Pins Detected = 12
Blinking LED on pin 13
1
2
3
4
5
6
7
8
9
10
PyMata close(): Calling sys.exit(0): Hope to see you soon!
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/site-packages/PyMata/pymata_serial.py", line 122, in run
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 450, in inWaiting
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'unpack'
```

zawilliams, 2015-03-26:
Ok weird - rebooted the Yun and there are no issues now. Everything is working. Closing this.

zawilliams, 2015-03-26:
After running the blink script again, it's still happening. Here's what I'm getting this time.

```
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/site-packages/PyMata/pymata_serial.py", line 122, in run
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 449, in inWaiting
<type 'exceptions.IOError'>: [Errno 9] Bad file descriptor
```

Any ideas?

MrYsLab, 2015-03-26:
A small note - the latest verson of PyMata is 2.06. This will not fix your problem, but there is a nasty bug if you send PyMata the reset firmata command. So when you get a chance you should probably update.

It looks like there may be a race condition between when the serial port is closed and when the serial read thread goes to retrieve the next character - the port is already closed and there is no port there.

This shouldn't happen, but it causes no harm other than the ugliness of the exception report.

Since I can't reproduce the problem here, I can send you a patched version of pymata_serial.py to see if figured out a potential fix. If you send an email  to MisterYsLab@gmail.com and I will be happy to reply with the patched file attached.

Also, are you using PyMata loaded in Linux on the Yun, or are you running PyMata from a PC connected to the Yun? If you are running PyMata on Yun Linux, which version of Linux are you using?

zawilliams, 2015-03-26:
Thanks @MrYsLab - Good to know about the latest version. I was pulling in the latest tarball from the /dist directory in the repo instead of pulling in the whole repo to save space on the Yun. I'll update to the latest version - somehow I missed that.

Gonna try the patch in pymata_serial.py and will report back.

I'm running PyMata on the Yun and using the OpenWRT-Yun version of Linux. It's based on OpenWRT Attitude Adjustment. Also running Python 2.7.

MrYsLab, 2015-03-26:
Is that the Release 1.5.3 - November 13th, 2014 version? In case the patch doesn't work, I want to test here to see if I can reproduce it.

Also, which version of Firmata are you using?
Thanks.

zawilliams, 2015-03-26:
Correct - 1.5.3 is what I'm using. I just tried the patch without upgrading to PyMata 2.06. Was still running 2.05 (I pulled down a new dist of 2.05, modified pymata_serial.py and then ran `python setup.py install`) and it didn't work. Still got this after the blinking counts to 10:

```
PyMata close(): Calling sys.exit(0): Hope to see you soon!
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/site-packages/PyMata/pymata_serial.py", line 123, in run
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 449, in inWaiting
<type 'exceptions.IOError'>: [Errno 9] Bad file descriptor
```

Should I try the upgrade to 2.06 and then apply the patch? Should the latest version be in the /dist directory as a tarball?

I am using a modified version of Firmata that works with ATH0 for the Yun. It was originally from here: https://gist.github.com/edgarsilva/e73c15a019396d6aaef2

I then replaced setup() with this version of setup(): http://forum.arduino.cc/index.php?topic=191820.msg1436262#msg1436262 but I replaced the baud rate with 57600 to match pymata_serial.py

The resulting Firmata sketch is here: https://gist.github.com/zawilliams/8a3dbaee4a06c4fa080e

Here is the blinking script: https://gist.github.com/zawilliams/d406fe67833f7b6293e4

MrYsLab, 2015-03-26:
Hi Zach,
     Both github and Pypi have version 2.06, so I am not sure why you 
are getting 2.05.

I wouldn't upgrade just yet, since if I can reproduce the problem here, 
then I will be uploading another version when I come up with a fix.

I will upgrade my Yun and get back to you on what I find. Hopefully that 
will be soon, but I will keep you updated.

Alan

zawilliams, 2015-03-26:
Hey Alan - I'm doing a wget from here: https://github.com/MrYsLab/PyMata/tree/master/dist and then unzipping the file and installing that way instead of downloading the whole repo. I only have around 8 megs to deal with on the Yun and I have other stuff installed taking up about 4 megs. That was the reason I was pulling in just that tarball in the /dist directory. I just pulled in what looked to be the latest instead of checking the repo's setup.py file for the latest version.

The latest version in the /dist is just the 2.06 so I'm assuming that's just an archive instead of having the latest version tarball.

Thanks for checking on that. Let me know if you need any other details. I'll get them to you as soon as I can.

MrYsLab, 2015-03-26:
Sorry about that. I just updated dist to have 2.06. Thanks for letting 
me know.

Alan

MrYsLab, 2015-03-27:
Zach,
    I used the latest OS update, the stock v2.06 version of PyMata, and your Arduino sketch and blink script, and it does not fail for me.
Did you comment out the line in /etc/inittab that deals with ttyATH0?

```
::sysinit:/etc/init.d/rcS S boot
::shutdown:/etc/init.d/rcS K shutdown
#ttyATH0::askfirst:/bin/ash --login
```

zawilliams, 2015-03-27:
Thanks for throwing 2.06 in there! No worries!

I did comment out the line in /etc/inittab

It didn't fail for me for the first few runs of the blink script, but after I ran it a few times (typically the 4th time it did it), then I ran into the exception. Gonna try to install 2.06 and see what I get.

zawilliams, 2015-03-27:
Alright, so I installed 2.06 and still am getting the same exception.

Since I'm installing 2.06 and building from source, I also needed to install pySerial - I installed v2.7.

Anything else you can think of?

MrYsLab, 2015-03-27:
That should be all you need. BTW, if you want to make installation easier, you can install pip on the yun. 
Instructions are here: http://codybonney.com/installing-pip-on-the-arduino-yun/

So if you are not seeing the exceptions each time you run the script , it is definitely a race condition. I uploaded a modified blink script that issues a firmata reset before the close : https://gist.github.com/MrYsLab/aabc3e7391ebc7ffd96d

Please let me know if this solves the problem.

zawilliams, 2015-03-28:
Hey Alan,

Tried the modified blink script and it's still happening. It doesn't seem to happen on the first run. Initially it seemed to do it after the first few runs, but it looks like it's happening after the 1st run of the script.

One thing I noticed is that upon booting the Yun, the serial Rx light is off. Once I start the script using `python blink.py` the serial Rx light comes on when the script opens the serial connection. Once the script is done, the Rx light stays on - I'm guessing the connection doesn't close? If I start the script again, it'll run through blinking but then gives the exception after the `PyMata close()` line.

Would the Rx light staying on indicate a potential problem with the serial connection not closing?

MrYsLab, 2015-03-28:
The lights behave totally differently on my Yun. When I power up the Yun, here is what I see:
1. Power light comes on (green and never extinguishes).
2. After several seconds, the Yellow Rx light comes on and never extinguishes.
3. After the device connects to my LAN the USB light (white or blue - not sure) comes on and never extinguishes. Hmm, I would have thought the WLAN Led would have come on - not sure what is going on here.

When running the script the only other light that toggles is L13.

Other than the aesthetics of getting the exception, does this cause any issues in what you are trying to do? The exception does state: "(most likely raised during interpreter shutdown)" and then Python exits.

Out of curiosity, have you tried loading StandardFirmata (from the 1.61 IDE) onto the Yun and then connect usb and run PyMata and the blink script from your PC? I would be curious to know if you still see the exceptions in that configuration.

zawilliams, 2015-03-28:
Other than aesthetics of the exception, everything seems to still be working just fine when I fire up the script.

I loaded StandardFirmata (1.61 IDE) onto the Yun and connected via USB and ran the blink script from my computer. Changed the instance to connection to "/dev/tty.usbmodem1411" and it worked with no exceptions. Ran it 5 times and no issues.

One thing that is interesting is that the Rx light only came on when it was receiving a command to turn on or off the light. I'm guessing that is because the version of Firmata I was using was doing a polling instead for the Rx. I'm thinking this is the issue here.

The thing is that the modified Firmata allowed "/dev/ttyATH0" so I could communicate from the Atheros to the MCU all on the Yun. I need to look at the differences and see what was changed in my modified Firmata to allow for this. Will check tonight.

MrYsLab, 2015-03-30:
@zawilliams  I am going go close this issue, since it seems to be happening only on a Yun when operating from OpenWRT and the exception is primarily cosmetic.

If you feel that I prematurely closed the issue, please reopen it.

Resolution: No code change. The problem happens occasionally only when using a Yun and running PyMata in OpenWRT linux. The exception reported happens upon exiting a python script and appears to be due to a thread closing race condition in OpenWRT. The code has been tested on a PC using Ubuntu Linux and in Windows 8.1 with no exceptions reported. Also running PyMata on either a BeagleBone Black or Raspberry Pi that is connected to an Arduino Leonardo or Uno shows no exceptions.

closed 2015-03-30 by MrYsLab

zawilliams, 2015-03-31:
@MrYsLab  - Sounds good. Thanks for your help on this Alan. Much appreciated.