MrYsLab/PyMata #16: Requirements

fabaff, 2015-06-22:
It seems that pyserial is not pulled-in automatically. A pylint check in travis complaint about it. 

Shouldn't 'install_requires' be present in the setup.py file for pyserial?

MrYsLab, 2015-06-22:
You are correct. I had had it in setup.py in earlier releases, but it did not pull in serial.py so I removed the line. I will investigate update setup.py if I can get it working.

MrYsLab, 2015-06-24:
I just created release 2.07a and pushed to pypi. Pyserial should now be pulled in when using pip. I had to change the version from numeric to string since it caused a syntax error in Python3.

closed 2015-06-24 by MrYsLab

fabaff, 2015-06-25:
I see another issue:
```bash
    /usr/lib/python3.4/site-packages/setuptools/dist.py:285: UserWarning: Normalizing '2.07' to '2.7'
      normalized_version,
```

MrYsLab, 2015-06-25:
What are you doing to generate the warning? Also, is this happening with 2.07a?

fabaff, 2015-06-25:
PyMata was added to requirements.txt, then
```
python3 -m pip install --user -r requirements.txt
```
In the user's site-packages folders are named like this
```
PyMata
PyMata-2.7-py3.4.egg-info
```
I don't know if this is related to pip 6.0.8 or Python 3.4.

MrYsLab, 2015-06-25:
I am using pip 7.0.3 and am not seeing the problem.
Here is the contents of my requirements file:
```
PyMata==2.07a
```
And the output of running the command:
```
sudo python3.4 -m pip install --user -r requirements.txt
The directory '/home/afy/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/afy/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting PyMata==2.07a (from -r requirements.txt (line 1))
  Downloading PyMata-2.07a.tar.gz
Requirement already satisfied (use --upgrade to upgrade): pyserial==2.7 in /usr/local/lib/python3.4/site-packages (from PyMata==2.07a->-r requirements.txt (line 1))
Installing collected packages: PyMata
  Running setup.py install for PyMata
Successfully installed PyMata
```

fabaff, 2015-06-25:
After playing around with different options, updating to pip 7.0.3, and using `PyMata==2.07a` instead of `PyMata>=2.07a` I don't see the warning anymore.

My guess is that pip 6.0.8 can work with chars in the version number. 

I would suggest to get rid off the zero in the version. Sooner or later somebody will run into trouble...I'm not sure but I think that RPM will preserve the zero while pip doesn't. And that would lead to multiple installed versions of PyMata. Just talking from a packager point of view.

Anyway, thanks for all your help to solve this.

fabaff, 2015-06-25:
Sorry, again...I just have learned  that `a` means `alpha` and that's the reason for `==`.

MrYsLab, 2015-06-25:
Thanks again. I just created version 2.08 and hopefully that will make things less confused.

Just to let you know, I am preparing the next generation of PyMata for release sometime this summer. It is asyncio based (Python >= 3.4.3). Code is complete and performance is excellent. The release will contain three APIs for the user to choose from. The first is pure asyncio. I am calling the two other APIs "plug-ins" and they use the pure asyncio API internally. One of the plug-ins looks like non-asyncio for those that do not want or need to use asyncio directly. The other plugin provides a websocket interface to allow users to connect to firmata via webpage. I am using the Autobahn websocket library which will allow me to add some additional features later when crossbar.io gets ported to asyncio (currently it uses twisted which I do not want to use). I hope to publish some information on my blogger blog in the next few days about this release. It will be published to  a whole new repository since it is not totally compatible with PyMata 2.08 and it won't support Python 2.7.