MrYsLab/PyMata #20: Send and receive string messages

murmlgrmpf, 2015-12-26:
Hi, thank you for this brilliant tool!

I am implementing an interface to send and receive json objects via serial connection between arduino and a linino openwrt implementation (arduino yun).
Therefore I need to send and receive string objects on both sides. On the arduino, this is straight forward, but on the pymata side, strings are only printed to the console for debug purposes.
I would suggest to implement a more general handler for strings and move the debugging code to pymata.

Shall I provide a sample implementation?

MrYsLab, 2015-12-26:
This sounds like application level coding and not something that normally would be part of the library, but please send a pointer to your work and I would be happy to take a look. If it makes sense, I would include it.

For pymata-aio I provide an application called [pymata_iot](https://github.com/MrYsLab/pymata-aio/wiki/pymata_iot-Usage-Notes) that provides a json translation of the entire Firmata protocol. I have also included a web page that connects to pymata_iot that allows you to control an Uno all from that web page.  

I am also using json messaging for this [project](https://github.com/MrYsLab/rb4s/blob/master/redbot_controller.py). It uses the pymata_core library (the next generation of the PyMata library).

murmlgrmpf, 2015-12-26:
You are right, I have a json interface implemented on the arduino microprocessor (mega2560). Firmata is too heavy for this purpose. I have altered the pymataSerial to call a callback function that fits my needs. As my interface is character based, this generates way less overhead.
Thank you for writing/maintaining pymata and pymataSerial!

closed 2015-12-26 by murmlgrmpf

MrYsLab, 2015-12-27:
@murmlgrmpf I am not sure if this will be any help, but a while ago Scratch used a JSON based interface. The [s2a](https://github.com/MrYsLab/S2A) project was implemented using JSON and not firmata. If you can make use of the sketch or any of the code, please do. One of the problems I had, was that my JSON scheme was very slow compared to firmata.

If you ever publish your project, please let me know, I would love to take a look.

murmlgrmpf, 2015-12-28:
@MrYsLab after having a look at your code for https://github.com/MrYsLab/S2A, I can tell, that in deed the solution is very similar to what I am implementing.
There is an efficient json implementation: https://github.com/bblanchon/ArduinoJson which I am using on the arduino side. On the linino side of the yun, I will use your serial class, (the one in pymata with threading) and a callback for the processing of the incoming messages.
My code is part of a project to control a heating system at my parents home: https://github.com/murmlgrmpf/Arduheat
The controller is running for over a year now, but it is not very well documented and the serial stuff is not there yet (currently it is using the arduino bridge).

MrYsLab, 2015-12-28:
@murmlgrmpf Very impressive!

murmlgrmpf, 2016-01-02:
Thank you! We got the json based serial communication working now. It is much faster than the yun bridge and fast enough for our logging purposes. Without the pymata example, I would never have gotten this far on the python side (threading!).
In the end the arduheat project probably remains a unique device, as it relies on too many special hardware and software features.
@MrYsLab I really appreciate your efforts to make programming with arduino hardware accessible to kids with the scratch interface!

MrYsLab, 2016-01-03:
That is great news!