MrYsLab/PyMata #21: Specified folder/zip file does not contain a valid library

owenchen, 2015-12-27:
Hi 
I try to download the zip file or try to use the arduino zip file in the ArduinoSketch directory. 
Then, in Arduino IDE to perform the following step:
1) Sketch > Include library > Add .zip library

However, I got error in opening zip file
Specified folder/zip file does not contain a valid library
Can you help me to find the problem?

The detail of message as follows:
Arduino: 1.6.7 (Mac OS X), Board: "LinkIt Smart 7688 Duo"

Specified folder/zip file does not contain a valid library
java.util.zip.ZipException: error in opening zip file
	at java.util.zip.ZipFile.open(Native Method)
	at java.util.zip.ZipFile.<init>(ZipFile.java:219)
	at java.util.zip.ZipFile.<init>(ZipFile.java:149)
	at java.util.zip.ZipFile.<init>(ZipFile.java:163)
	at processing.app.tools.ZipDeflater.<init>(ZipDeflater.java:26)
	at processing.app.Base.handleAddLibrary(Base.java:2315)
	at processing.app.Base$7.actionPerformed(Base.java:1157)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:125)
	at java.awt.MenuItem.processActionEvent(MenuItem.java:669)
	at java.awt.MenuItem.processEvent(MenuItem.java:628)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:761)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
error in opening zip file
Specified folder/zip file does not contain a valid library
Specified folder/zip file does not contain a valid library

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

MrYsLab, 2015-12-27:
@owenchen I apologize for the confusion on installation. I have created a [wiki page](https://github.com/MrYsLab/pymata-aio/wiki/Uploading-FirmataPlus-to-Arduino with installation instructions.

I do not use the "normal" IDE methods for installation and that is why things failed. To summarize the installation, you  unzip the file directly into the Arduino sketch folder and no additional installation is necessary.

The reason for using this unorthodox method is that FirmataPlus has several other libraries as dependencies. Some of these libraries had to be modified as documented by their creators, and for some of my users, it became overly complicated, so I provide all of the libraries and their modifications.

If you have any other issues or questions, please let me know.

I am closing this issue, but you may reopen it if you continue to have issues.

closed 2015-12-27 by MrYsLab

owenchen, 2015-12-28:
Thank you, I can find the library in my IDE now.

MrYsLab, 2016-10-05:
On my system the Arduino IDE installs external libraries in a directory called Arduino.  On my Windows machine, it looks like this:

```
C:\Users\Alan\Documents\Arduino\libraries>dir /w
 Volume in drive C has no label.
 Volume Serial Number is 5470-4840

 Directory of C:\Users\Alan\Documents\Arduino\libraries

[.]                 [..]                [AdaEncoder]        [cbiface]
[cppfix]            [EnableInterrupt]   [Firmata]           [FirmataPlus]
[FirmataPlusLBT]    [FirmataPlusRB]     [FirmataPlusRBPixy] [NewPing]
[ooPinChangeInt]    [Pixy]              readme.txt          [RedBot]
[RedBot2]           [Temboo]
               1 File(s)             87 bytes
              17 Dir(s)  22,272,012,288 bytes free
```

and on Linux it looks like this:
```
$ :~/Arduino/libraries$ ls
AdaEncoder       Firmata          FirmataPlusRB      Pixy
cbiface          FirmataPlus      FirmataPlusRBPixy  readme.txt
cppfix           FirmataPlus32u4  NewPing            RedBot
EnableInterrupt  FirmataPlusLBT   ooPinChangeInt     RedBot2
```

If you have a different directory structure, then please install as shown above. If you directory structure looks like what I show above and you are still having issues, please let me know.