MrYsLab/PyMata #22: Arduino due

vinny74man, 2016-02-07:
Hi, I have an "Arduino due" and until now I used "pyfirmata" mo I would try "PyMata" I just can not figure out how to identify the board as a "due" because if you use the pin 13 as PWM I say that is not possible, since the pin 13 is not PWM, which is true for a "arduino one" but not for my.
Thank you
P.S. This is an on-line translation

MrYsLab, 2016-02-07:
Hi. Could you please provide some clarification:
1. Are you using a Due or an Uno?
2. If you are using a Due, what PyMata command are you using that is saying that Pin 13 is not PWM?
The determination of Pin type is provided in the Firmata boards.h file. PyMata just reads that file.

vinny74man, 2016-02-08:
Hi, I use "arduino due"

Thank you

P.S. This is an on-line translation

vinny74man, 2016-02-08:
From a sample taken in the network command is this,

"# Set the white LEDs for PWM operation
firmata.set_pin_mode (WHITE_LED, firmata.PWM, firmata.DIGITAL) "

if "WHITE_LED" = 6 the program progresses,
if "WHITE_LED" = 13 the program stops,

P.S. This is an on-line translation

MrYsLab, 2016-02-08:
I just tried this program on a DUE clone and it works. Could you please try this and let me know if it works for you?

```
import time
from PyMata.pymata import PyMata

# Test pymata
def pymata_test():
    print ("Testing PyMata ...")
    # create a PyMata instance
    # set the COM port string specifically for your platform
    firmata = PyMata("/dev/ttyACM0")

    # set pin 13 to PWM mode
    firmata.set_pin_mode(13, firmata.PWM, firmata.DIGITAL)

    # set LED to PWM level 10

    firmata.analog_write(13, 10)
    print('Dim')

    time.sleep(2)

    # set LED to full brightness
    firmata.analog_write(13, 100)
    print('Bright')

    time.sleep(2)

    firmata.analog_write(13, 255)
    print('Brightest')

    firmata.close()

if __name__ == "__main__":
    pymata_test()

```

vinny74man, 2016-02-08:
Ok thanks, I try

P.S. This is an on-line translation

vinny74man, 2016-02-09:
ok thanks, it recognizes the "Arduino Due" but the sketch goes only once after the pin is at the high state and until the reconfigured as a digital pin is always on.

I loaded on board the sketch "SStandard Firmata Plus" and "Standard Firmata" with the same result in the release of "Firmata" is the 2.5.1

P.S. This is an on-line translation

MrYsLab, 2016-02-09:
Here is another test program. I ran this program on an Uno with an LED on pin6, then on the DUE with an LED on PIN 12 and lastly on the DUE using PIN13 both with an external LED attached and just using the internal LED.

The program queries the pin state at the Firmata sketch, so the data is coming from StandardFirmata. The 3 values shown in the output below are [pin number, pin mode, pin value]

pin modes are as following: 0 = input, 1 = output and 3 = pwm

As you can see from the outputs below, there is no differences other than the pin numbers.

When I connect an external LED to PIN 13 of the DUE the LED is always off. For pin 12 this does not happen. I can only conclude that the DUE is pulling down this pin to handle the internal LED and that the issue is a DUE issue and not PyMata, and therefore I am closing this issue.

PROGRAM

```
import time
from PyMata.pymata import PyMata

# Test pymata
def pymata_test():
    
    # change this value to change pin under test
    pin_under_test = 6

    print ("Testing PyMata ...")
    # create a PyMata instance
    # set the COM port string specifically for your platform
    firmata = PyMata("/dev/ttyACM0")

    firmata.pin_state_query(pin_under_test)
    time.sleep(1)

    print('pin state after creating object: ' + str(firmata.get_pin_state_query_results()))

    # set pin pin_under_test to PWM mode
    firmata.set_pin_mode(pin_under_test, firmata.PWM, firmata.DIGITAL)
    time.sleep(1)

    firmata.pin_state_query(pin_under_test)

    time.sleep(1)

    print('pin state after setting mode to PWM: ' + str(firmata.get_pin_state_query_results()))

    # set LED to PWM level 10

    firmata.analog_write(pin_under_test, 10)
    # print('Dim')

    firmata.pin_state_query(pin_under_test)
    time.sleep(1)

    print('pin state after setting pin value to 10: ' + str(firmata.get_pin_state_query_results()))

    # set LED to full brightness
    firmata.analog_write(pin_under_test, 100)

    firmata.pin_state_query(pin_under_test)

    firmata.pin_state_query(pin_under_test)
    time.sleep(1)

    print('pin state after setting pin value to 100: ' + str(firmata.get_pin_state_query_results()))

    time.sleep(2)

    print('pin state before setting mode to OUTPUT: ' + str(firmata.get_pin_state_query_results()))

    firmata.set_pin_mode(pin_under_test, firmata.OUTPUT, firmata.DIGITAL)

    firmata.digital_write(pin_under_test, 0)

    firmata.pin_state_query(pin_under_test)

    time.sleep(1)

    print('pin state after setting mode to OUTPUT and setting pin to 0: ' + str(firmata.get_pin_state_query_results()))

    firmata.digital_write(pin_under_test, 1)

    firmata.pin_state_query(pin_under_test)

    time.sleep(1)

    print('pin state after setting mode to OUTPUT and setting pin to 1: ' + str(firmata.get_pin_state_query_results()))

    time.sleep(2)

    firmata.digital_write(pin_under_test, 0)

    firmata.pin_state_query(pin_under_test)

    time.sleep(1)

    print('pin state after setting mode to OUTPUT and setting pin to 0: ' + str(firmata.get_pin_state_query_results()))

    time.sleep(2)

    firmata.close()

if __name__ == "__main__":
    pymata_test()
```

OUTPUT FROM RUNNING ON UNO PIN 6
```
PyMata version 2.11  Copyright(C) 2013-15 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyACM0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 2 seconds
Total Number of Pins Detected = 20
Total Number of Analog Pins Detected = 6
pin state after creating object: [6, 0, 0]
pin state after setting mode to PWM: [6, 3, 0]
pin state after setting pin value to 10: [6, 3, 10]
pin state after setting pin value to 100: [6, 3, 100]
pin state before setting mode to OUTPUT: []
pin state after setting mode to OUTPUT and setting pin to 0: [6, 1, 0]
pin state after setting mode to OUTPUT and setting pin to 1: [6, 1, 1]
pin state after setting mode to OUTPUT and setting pin to 0: [6, 1, 0]
PyMata close(): Calling sys.exit(0): Hope to see you soon!

Process finished with exit code 0
```

OUTPUT FROM RUNNING ON DUE PIN 12
```
PyMata version 2.11  Copyright(C) 2013-15 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyACM0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 66
Total Number of Analog Pins Detected = 12
pin state after creating object: [12, 1, 0]
pin state after setting mode to PWM: [12, 3, 0]
pin state after setting pin value to 10: [12, 3, 10]
pin state after setting pin value to 100: [12, 3, 100]
pin state before setting mode to OUTPUT: []
pin state after setting mode to OUTPUT and setting pin to 0: [12, 1, 0]
pin state after setting mode to OUTPUT and setting pin to 1: [12, 1, 1]
pin state after setting mode to OUTPUT and setting pin to 0: [12, 1, 0]
PyMata close(): Calling sys.exit(0): Hope to see you soon!

Process finished with exit code 0
```

OUTPUT FROM RUNNING ON DUE PIN 13 - There is no change in output if an external LED is used
```
PyMata version 2.11  Copyright(C) 2013-15 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyACM0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 66
Total Number of Analog Pins Detected = 12
pin state after creating object: [13, 1, 0]
pin state after setting mode to PWM: [13, 3, 0]
pin state after setting pin value to 10: [13, 3, 10]
pin state after setting pin value to 100: [13, 3, 100]
pin state before setting mode to OUTPUT: []
pin state after setting mode to OUTPUT and setting pin to 0: [13, 1, 0]
pin state after setting mode to OUTPUT and setting pin to 1: [13, 1, 1]
pin state after setting mode to OUTPUT and setting pin to 0: [13, 1, 0]
PyMata close(): Calling sys.exit(0): Hope to see you soon!

Process finished with exit code 0
```

closed 2016-02-09 by MrYsLab

MrYsLab, 2016-02-09:
Additional information. I ran the following sketch with an external LED on pin 13 and it also does not light the LED. 
This sketch does not use firmata or PyMata.

```
void setup() {
  // put your setup code here, to run once:
  pinMode(13, OUTPUT);

}

void loop() {
  // put your main code here, to run repeatedly:
  analogWrite(13, 200);

}
```