MrYsLab/PyMata #23: Problem reading ping with multiple Arduino devices

kwunder, 2016-02-13:
Hi, I have two Arduino boards connected simultaneously and have a problem with reading from (digital) input pins. 

I do something like the following:
`a1=PyMata("/dev/ttyACM0")
a2=PyMata("/dev/ttyACM1")
a1.set_pin_mode(11,a1.INPUT,a1.DIGITAL)
a1.digital_write(11,1)
a2.set_pin_mode(11,a2.INPUT,a2.DIGITAL)
a2.digital_write(11,1)
while 1:
   print(str(a1.digital_read(11)) + " " + str(a2.digital_read(11)))
`
When I now connect GND to pin 11 of either device, then the program will print "0 0", so both a1 and a2 return the same value. 

This problem occurs only when I read values from input ports. Writing to output ports individually for each device works fine.

MrYsLab, 2016-02-13:
@kwunder I think your problem is that you are trying to control the pin value with the digital write. Firmata is rather limiting. If you set a pin to be an input, performing a write has no effect. (That is not the case for a "normal" arduino sketch - doing a write on a "normal" sketch does work).

This is not related to the number of Arduinos. The same problem will happen when using a single Arduino.

After remove the digital writes from your program, try wiring up a something like a push button as an inputs to each board. You should see the values change independently for both boards.

I am closing this issue. If you are still having issues or want to add a comment you can after it is closed. If there is still an issue, I will reopen it.

Thanks,
Alan

kwunder, 2016-02-13:
Hi Alan,

the digital_write is just to enable the pullup resistor and my issue has nothing to do with it. The problem is that the digital_response_table in the PyMataCommandHandler is always updated to identical values for both a1 and a2 regardless of whether I change an input pin on board1 or board2.

MrYsLab, 2016-02-13:
Ok. I am reopening this. I will try to recreate the problem.

MrYsLab, 2016-02-13:
@kwunder I have confirmed the issue, but do not have an explanation as to why it is happening as of yet. Just to let you know, I tested this using pymata-aio and the issue does not occur. If you can go to python 3.5, I would suggest you use pymata-aio. 

I will update when I have an answer.

MrYsLab, 2016-02-13:
@kwunder The PyMata threading model was designed to handle a single instance per process. What you wish to accomplish is a perfectly legitimate request, but would mean redesigning the entire library. Since pymata-aio does do what you require, the solution would be to use the pymata-aio library.

Here is a sample of [pymata-aio code](https://gist.github.com/MrYsLab/4ed6c58755039d67881c) that accomplishes what you are trying to do.

If you do move over to pymata-aio, you will need to either run the StandardFirmata sketch or the pymata-aio version of FirmataPlus (it is different from the PyMata version). 

Please let me know if you have any questions.

Thanks,
Alan

closed 2016-02-13 by MrYsLab