MrYsLab/PyMata #25: PyMata doesn't support CH341PAR / CH341SER

nenodias, 2016-03-29:
It's a chinese board, the auto discover fail.
Have one solution?
Thanks

MrYsLab, 2016-03-29:
I am not familiar with CH341PAR / CH341SER. If this is a 100% compatible arduino clone , you will need to specify the com port when instantiating PyMata. PyMata does not perform auto port discovery. An example:

```
from PyMata.pymata import PyMata

# create a PyMata instance
board = PyMata("/dev/ttyACM0")
 ```

Could you please provide an example of your code and the error you are seeing.

nenodias, 2016-03-29:
`import sys
import signal
import PyMata.pymata
from PyMata.pymata import PyMata
# followed by another imports your application requires

# create a PyMata instance
# set the COM port string specifically for your platform
board = PyMata("/dev/ttyUSB0")

# signal handler function called when Control-C occurs
def signal_handler(signal, frame):
    print('You pressed Ctrl+C!!!!')
    if board != None:
        board.reset()
    sys.exit(0)

# listen for SIGINT
signal.signal(signal.SIGINT, signal_handler)

# Your Application Continues Below This Point`
Here is the code I was tested.
The out message is "PyMata version 2.12  Copyright(C) 2013-16 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyUSB0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board Auto Discovery Failed!, Shutting Down".

I tryied with https://github.com/thearn/Python-Arduino-Command-API and was worked.
Thanks for reply

MrYsLab, 2016-03-29:
Did you upload StandardFirmata to your board? The sketch from the link above will not work with PyMata.

nenodias, 2016-03-29:
No..., I will search how that work's.
Thanks for reply.
I'm a beginner with arduino xD

MrYsLab, 2016-03-29:
No problem. Here is a link that might help: https://github.com/MrYsLab/pymata-aio/wiki/Uploading-StandardFirmata-To-Arduino

MrYsLab, 2016-03-30:
I am closing this issue. If you have any additional questions, please either leave a comment here or open a new issue.

closed 2016-03-30 by MrYsLab