MrYsLab/PyMata #26: Moved pymata-aio issue #44 here

MrYsLab, 2016-09-03:
An issue  https://github.com/MrYsLab/pymata-aio/issues/44 should have been issued against PyMata and not pymata-aio. This issue is moved and documented here. PyMata constant values did not match the latest versions of FirmataPlus and caused Sonar support to work incorrectly.

This was fixed in version 2.13

rricardi, 2016-09-03:
The first code continuous  with the same error:
```

Python Version 2.7.10 (default, Oct 14 2015, 16:09:02) 
[GCC 5.2.1 20151010]

PyMata version 2.13  Copyright(C) 2013-16 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyUSB0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 1 seconds
Total Number of Pins Detected = 22
Total Number of Analog Pins Detected = 8
Traceback (most recent call last):
  File "hc-sr04.py", line 46, in <module>
    print(str(data[2]) + ' centimeters')
KeyError: 2

```
But the other code worked :smile: 

Now we make sure that the sensor works, right? what is the problem then to not work on Scratch? :(

rricardi, 2016-09-03:
Do you have some code in Scratch that simulates this code you gave me that you know is working certainty?

MrYsLab, 2016-09-03:
I should have been more clear. I tested using the same Scratch program you provided me. It worked without issue. When you start s2a_fm, make sure that it says PyMata v.213.

rricardi, 2016-09-03:
Worked now!!!!!! I was using this interface to communicate:
https://github.com/ComputacaoNaEscola/Scratchduino
but it does not use the version of Pymata you spoke.
Thank you!!!

rricardi, 2016-09-03:
This interface use PyMata version 1.57

MrYsLab, 2016-09-03:
I am glad that it works now. Thank you for reporting this. I am closing this issue since it appears to be solved.

closed 2016-09-03 by MrYsLab