MrYsLab/PyMata #28: PyMata and inifinte loop

acca84, 2016-10-19:
Hello,

I'd like to use pymata to read values from analog port of arduino and display it on a screen using a raspberry pi and pygame. My python code looks like the following : 

```
if __name__ == '__main__':
	

	val1=board.analog_read(1)/1000
	val2=board.analog_read(2)/1000
	sleep(0.5)
	threading.Thread(target=affiche_pygame).start()		

	while (lecture):
		for event in pygame.event.get():
			if event.type==QUIT:
					
				lecture=False	
				sleep(0.5)
				pygame.quit()		
				board.close()
				
				quit()
				sys.exit
			if event.type==pygame.locals.KEYDOWN:
				
				lecture=False
				sleep(0.5)
				pygame.quit()
				board.close()
				sleep(0.5)
				quit()
				sys.exit

		val1=board.analog_read(1)/1000
		val2=board.analog_read(2)/1000
		sleep(0.1)
		
		print val1
```

The thread "affiche_pygame" is an infinite loop wich takes val1 and val2 values and draw a rectangle wich length is calculated using val1 and val2
it works well on a raspberry pi B+
But on a raspberry pi 2 wich is faster, i can get first analog values then values are not updated, i can see the Arduino Tx light stop blinking after approx 10second.

If i add a sleep(0.1) at the end of the "affiche_pygame" while loop, it works but my animation is not smooth.

Do i stress pymata too much ?
If i don't launch the "affiche_pygame" thread, it works well

How can i have updated values without slowing my loop please ? I only want the latest up to date analog value of analog1 and analog2

I hope it's clear, sorry for my poor english.

Regards

I have tested pyfirmata lib and the same problem occurs

acca84, 2016-10-19:
i think there is a problem with pygame, because if i remove pygame.display.update() in my  "affiche_pygame" function, arduino continues sending data. Memory leak ?

MrYsLab, 2016-10-20:
I am not familiar with pygame or its event loop, but I suspect that the problem is integrating your code with the pygame event loop. I doubt if there is a memory leak causing your issues. I solved a similar situation for a tkinter gui I wrote for a pymata_aio project and the solution is documented  [here](https://github.com/MrYsLab/xibot/wiki/Integrating-XideKit-Components-with-External-Event-Loops).  Perhaps this may lead to a solution for your problem.

Since this does not appear to be a pymata issue I am closing this issue. If you feel it should remain open or would like to discuss this further, please feel free to contact me.

closed 2016-10-20 by MrYsLab

acca84, 2016-10-21:
ok thanks

I think i have found the problem.
Not sure but i think it was a global variable problem