MrYsLab/PyMata #29: Compatibility and Python 3.6

NervenCid, 2017-01-12:
Hello:

I'm new using firmata protocol, and looking a first steps guide...

pyFirmata was installed well aparantely in Python 3.6... is recommendable to use it on Python 3.6?

This library is compatible with all arduino boards?

Thank you

MrYsLab, 2017-01-12:
Python 3.6 should work fine. The firmataplus libraries are based off StandardFirmata version 2.5 and so should support the boards that StandardFirmata supports. It works with any of the Atmel processors such as those used in the Uno, Leonardo, Mega2560, and Teensy to name a few.

MrYsLab, 2017-01-13:
I am closing this since it was a question and not an issue. If you have any other please feel free to ask.

closed 2017-01-13 by MrYsLab