MrYsLab/PyMata #30: noob questions

ash73, 2017-02-05:
1. is there a version of PyMata for Python 3 on raspberry pi?

2. how do I set the analogue outputs on pins 3,5,6,9,10,11?

TIA.

MrYsLab, 2017-02-05:
You will need to install the StandardFirmata or FirmataPlus sketch on your Arduino.
Next install PyMata on the Raspberry Pi. If you want to use it with Python 3, then in a command window type:

```
sudo pip3 install PyMata
```
Below is a typical script to read a potentiometer connected to A2 on the Arduino. You would need to initialize all the other pins in a similar fashion and do reads on the specific pin.

```
import time
from PyMata.pymata import PyMata

# Create a PyMata instance
board = PyMata("/dev/ttyACM0", verbose=True)

# Analog pin
POTENTIOMETER = 2

# Set pin mode
board.set_pin_mode(POTENTIOMETER, board.INPUT, board.ANALOG)

while True:
    analog = board.analog_read(POTENTIOMETER)
    time.sleep(.2)
```
Let me know if you have any other questions.

ash73, 2017-02-05:
Thanks that's very helpful.

Is there a way to set the PWM pins (value 0-255)? Something like analog_write?

MrYsLab, 2017-02-05:
You may wish to view the documentation for the entire API here: http://htmlpreview.github.io/?https://github.com/MrYsLab/PyMata/blob/master/documentation/html/PyMata.pymata.PyMata-class.html

Here is a program that will fade the LED on pin 13 up and down:

```
import time
from PyMata.pymata import PyMata

# Create a PyMata instance
board = PyMata("/dev/ttyACM0", verbose=True)

# PWM Pin 
LED = 13

# Set pin mode
board.set_pin_mode(LED, board.PWM, board.DIGITAL)

analog_value = 0
while True:
    board.analog_write(LED, analog_value)
    analog_value += 1
    if analog_value == 256:
        analog_value = 0
    time.sleep(.2)
```

ash73, 2017-02-05:
Thanks for your help!

I'm testing it by reading a potentiometer on one pin and sending it as a digital output to a LED on another pin, works ok but occasionally I get "IndexError: list index out of range" when initialising contact with the Arduino. Only seems to happen with certain pin combos, adding a delay after the read seems to help a bit.

I assume I can use any analogue input and any digital output?

MrYsLab, 2017-02-05:
There are issues with performing a software reset on some models of the Arduino's. If you restart your python script, the Arduino may be unaware of this, even though PyMata sends a reset commend. As a result, the Arduino keeps streaming data (usually analog) and when the python script comes on line, it is receiving streaming data it does not expect..

If the time delay works, use it, or reset the board after stopping the Python script by pulling out the USB cable, and plugging it back in, waiting for the LEDs on the Arduino to stop blinking and then restart your script.

The pymata-aio library has some added features to prevent this. In FirmataPlus (not part of StandardFirmata), there is a way to set a keep-alive timer. What this does is, if the Arduino does not receive a keep-alive poll within a certain period of time (the time is settable by the user), it forces a hardware reset on the Arduino. So if you kill your Python script, the keep_alive polls will no longer be sent, and the FirmataPlus sketch forces the board to go into a hardware reset. Streaming data is killed and everyone is happy.

If you continue to have issues with your script, please provide a copy and I will take a look at it.

MrYsLab, 2017-02-06:
I am closing this issue. If you have any other questions, you can add them here.

closed 2017-02-06 by MrYsLab

taleravidhi, 2017-06-14:
How do I install PyMata on my beagle bone board? 
I'm trying to clone the git repo https://github.com/MrYsLab/PyMata on the board but nothing happens. It just shows Cloning into 'PyMata'...

MrYsLab, 2017-06-14:
@taleravidhi 
If your beaglebone black is connected to the internet, the easiest way is by using pip.
First make sure that pip is installed on the BBB. If you open a console window or log into the BBB using ssh, type:
`
which pip
`
If you see a path a appear on your screen, with something like this:
`
/usr/local/bin/pip
`
You have pip installed, if not, you will need to install it. To do so, type:
`
sudo apt-get install python-pip
`
When that completes try the two steps above.

Next, to install PyMata, type:
`
pip install PyMata
`
This will install PyMata and all of its dependencies (pyserial) and you should be all set.

Just to be clear, PyMata will not control the GPIO pins on the BBB, but only on an Arduino. You can use the BBB to control an Arduino. If you need to control the GPIO pins on the BBB, you might want to look at this article: 
[https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/installation-on-ubuntu](url)

Please let me know if any of this is unclear or if you have any other questions.

taleravidhi, 2017-06-15:
Thank you @MrYsLab . My problem is solved by another way. I transferred the PyMata-master folder on the board using sftp and then installed PyMata directly by running pymata.py file.

MrYsLab, 2017-06-15:
I am glad you got it working.