MrYsLab/PyMata #31: Interfacing with BH1750

yangmillstheory, 2017-03-08:
Hi, very aware that this might be a usage problem on my end.

I'm trying to interface with the BH1750 sensor, and doing:

```python
#!/usr/bin/env python

import time
from PyMata.pymata import PyMata
import os

port = PyMata(os.environ['ARDUINO_PORT'])
port.i2c_config(0, port.ANALOG, 4, 5)

try:
  while True:
    port.i2c_read(0x23, 0, 2, port.I2C_READ)
    time.sleep(3)  # wait for data to be sent

    data = port.i2c_get_read_data(0x23)
    print 'got data', data
    lux = (data[1] << 8 | data[2]) >> 4
    lux /= 1.2
    print str(lux) + ' lux'
except KeyboardInterrupt:
    port.close()
```

I get the following output in the console:

```
(arduino-playground) arduino-playground git:(master) ✗ ./src/bh1750_sensor.py

Python Version 2.7.10 (default, Jul 30 2016, 19:40:32)
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.34)]

PyMata version 2.13  Copyright(C) 2013-16 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/cu.usbmodem1421

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 20
Total Number of Analog Pins Detected = 6
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
got data [0, 0, 11]
0.0 lux
^CPyMata close(): Calling sys.exit(0): Hope to see you soon!
```

My connections are set up as inhttps://www.mysensors.org/build/light-bh1750 with an Arduino UNO, and interfacing directly via sketch uploaded through Arduino IDE works (I can see the lux values change in the serial monitor of the IDE).

Do you have any thoughts as to what I'm doing wrong?

Thanks.

yangmillstheory, 2017-03-08:
The `11` is immaterial by the way, when I re-uploaded the sketch, tested the sensor (successfully), then ran the Python code above  again, I consistently get `3.33333 lux`.

MrYsLab, 2017-03-08:
The device you are using is an i2c device and not an analog device. This is from the Uno :product page:
TWI: A4 or SDA pin and A5 or SCL pin. Support TWI communication using the Wire library.
T
his library has an i2c sketch for the device:
https://github.com/claws/BH1750

You will need to convert that sketch into a PyMata i2c application. I have not used the BH1750, but I have implemented an application for the TMP102 temperature sensor:
https://github.com/MrYsLab/PyMata/blob/master/examples/i2c/pymata_i2c_read/i2c_read_polled.py

The 2 devices are different, and you will need to customize for the BH1750 specifics, but the TMP102application will give you an idea of what is involved.

yangmillstheory, 2017-03-08:
Thanks for the pointers Alan~

closed 2017-03-08 by yangmillstheory

yangmillstheory, 2017-03-09:
Wait, the code snippet above I posted is actually a PyMata i2c application.

MrYsLab, 2017-03-09:
Sorry about that - I need to wait to reply until after I have had my coffee ;-).

It appears that you need to first set the device's mode before reading. 
Try adding
```
        port.firmata.i2c_write(0x23, 16)
```
after your call  to i2c_config.

The modes are defined here if you want to view the options:
https://github.com/mysensors/MySensorsArduinoExamples/blob/master/libraries/BH1750/BH1750.h

yangmillstheory, 2017-03-12:
Still can't get this to work. The wiring is correct because I can get updated readings with the following sketch:

```sketch
/*

Example of BH1750 library usage.

This example initalises the BH1750 object using the default
high resolution mode and then makes a light level reading every second.

Connection:
 VCC-5v
 GND-GND
 SCL-SCL(analog pin 5)
 SDA-SDA(analog pin 4)
 ADD-NC or GND

*/

#include <Wire.h>
#include <BH1750.h>

BH1750 lightMeter;

void setup(){
  Serial.begin(9600);
  lightMeter.begin();
  Serial.println("Running...");
}

void loop() {
  uint16_t lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  delay(1000);
}
```

I get `0.0` consistently with the following PyMata application though:

```
#!/usr/bin/env python
from PyMata.pymata import PyMata
import os
import time

board = PyMata(os.environ['ARDUINO_PORT'])
board.i2c_config(0, board.ANALOG, 4, 5)
board.i2c_write(0x23, board.I2C_READ_CONTINUOUSLY) # same results with board.I2C_READ

while True:
  try:
    board.i2c_read(0x23, 0, 2, board.I2C_READ) # same results with board.I2C_READ_CONTINUOUSLY
    time.sleep(3)

    data = board.i2c_get_read_data(0x23)
    print 'Got read data: %s' % data
    lux = (data[1] << 8 | data[2]) >> 4
    lux /= 1.2
    print str(lux) + ' lux'
  except KeyboardInterrupt:
    board.close()
```

MrYsLab, 2017-03-12:
I just found this: https://books.google.com/books?id=O0PfBgAAQBAJ&pg=PA121&lpg=PA121&dq=firmata+bh1750&source=bl&ots=iXDiR1y27i&sig=8AuRNL3sUcmIZXsnvlzHbu-sip0&hl=en&sa=X&ved=0ahUKEwjW852T18_SAhVI3IMKHV5ZCnkQ6AEIMTAC#v=onepage&q=firmata%20bh1750&f=false

Please let me know if this works.

yangmillstheory, 2017-03-12:
Ah, that's actually what got me here in the first place. I'm reading through that book, and couldn't make that example work. Notice that the bottom of that snippet should `NameError`, which makes me question its integrity.

MrYsLab, 2017-03-12:
I am not sure what you are referring when you say NameError.

Nonetheless, I took a look at the DFRobot code: https://www.dfrobot.com/wiki/index.php/Bh1750#Sample_Code

Looking at that sketch, in the loop, the device is initialized by writing 0x10 to the chip. Then it delays for 200 ms. Next it performs a 2 byte read, and then delay for 150 ms, and then the loop starts again.

Assuming that that sketch works (I have no way of knowing), the PyMata code would look like:
```
#!/usr/bin/env python
from PyMata.pymata import PyMata
import os
import time

board = PyMata(os.environ['ARDUINO_PORT'])
board.i2c_config(0, board.ANALOG, 4, 5)

while True:
  try:
    board.i2c_write(0x23, board.I2C_READ_CONTINUOUSLY) # same results with board.I2C_READ
    sleep(0.2)
    board.i2c_read(0x23, 0, 2, board.I2C_READ) # same results with board.I2C_READ_CONTINUOUSLY
    time.sleep(3)

    data = board.i2c_get_read_data(0x23)
    print 'Got read data: %s' % data
    lux = (data[1] << 8 | data[2]) >> 4
    lux /= 1.2
    print str(lux) + ' lux'
  except KeyboardInterrupt:
    board.close()
```
Give that a try.

yangmillstheory, 2017-03-12:
For the `NameError` comment:

```
(arduino-playground) arduino-playground git:(master) ✗ python -c "name_that_doesnt_exist"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'name_that_doesnt_exist' is not defined
```

In the case of the book, it would be the reference to the nonexistent variable `firmata` on the last line. 

I'll try the snippet you posted above, thanks!

MrYsLab, 2017-03-12:
I just ordered a BH1750 and should receive it later this week. I will publish the code when I am able to get the device to respond.

yangmillstheory, 2017-03-12:
Thanks. I just ordered another one myself; for some reason I can't get the test sketch to run on my Uno (other example sketches run fine, though).

MrYsLab, 2017-03-18:
@yangmillstheory I haven't forgotten about you. The part never arrived from Amazon, so I just reordered and hopefully I will received it in 2 days.

MrYsLab, 2017-03-18:
@yangmillstheory 
The bh1750 finally came! I will first show the output from the program and the program will be below that:

```
$ python b.py

Python Version 2.7.12+ (default, Sep 17 2016, 12:08:02) 
[GCC 6.2.0 20160914]

PyMata version 2.13  Copyright(C) 2013-16 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyACM0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 30
Total Number of Analog Pins Detected = 12
Got read data: [0, 0, 182]
9.16666666667 lux
Got read data: [0, 0, 182]
9.16666666667 lux
Got read data: [0, 0, 182]
9.16666666667 lux
Got read data: [0, 0, 183]
9.16666666667 lux
Got read data: [0, 0, 182]
9.16666666667 lux
Got read data: [0, 0, 112]
5.83333333333 lux
Got read data: [0, 0, 40]
1.66666666667 lux
Got read data: [0, 0, 42]
1.66666666667 lux
Got read data: [0, 0, 3]
0.0 lux
Got read data: [0, 0, 42]
1.66666666667 lux
Got read data: [0, 0, 80]
4.16666666667 lux
Got read data: [0, 0, 58]
2.5 lux
Got read data: [0, 0, 39]
1.66666666667 lux
Got read data: [0, 0, 39]
1.66666666667 lux
Got read data: [0, 0, 37]
1.66666666667 lux
Got read data: [0, 0, 33]
1.66666666667 lux
Got read data: [0, 0, 182]
9.16666666667 lux
Got read data: [0, 0, 180]
9.16666666667 lux
Got read data: [0, 0, 181]
9.16666666667 lux
Got read data: [0, 0, 180]
9.16666666667 lux
^CPyMata close(): Calling sys.exit(0): Hope to see you soon!

```

I moved my hand around the device which caused the exepcted variances.

Now here is the script I used. Please note that I tested on a Leonardo. Just comment out the line for configuring on the leonardo and uncomment out the line for the uno, if you are using an uno.

```
from PyMata.pymata import PyMata
import os
import time

board = PyMata("/dev/ttyACM0")

# for leonardo
board.i2c_config(0, board.DIGITAL, 3, 2)

# for uno
# board.i2c_config(0, board.ANALOG, 4, 5)

while True:
  try:
    board.i2c_write(0x23, board.I2C_READ_CONTINUOUSLY) # same results with board.I2C_READ
    time.sleep(.2)
    board.i2c_read(0x23, 0, 2, board.I2C_READ) # same results with board.I2C_READ_CONTINUOUSLY
    time.sleep(.3)

    data = board.i2c_get_read_data(0x23)
    print 'Got read data: %s' % data
    lux = (data[1] << 8 | data[2]) >> 4
    lux /= 1.2
    print str(lux) + ' lux'
  except KeyboardInterrupt:
    board.close()

```

yangmillstheory, 2017-03-18:
Great! 

I'll verify this in a bit, as my BH1750 came as well. I must have wired them up incorrectly, or messed up soldering the header onto the board. 

Did you have to do any soldering yourself? Which board did you get?

MrYsLab, 2017-03-18:
Yes I had to solder the header on. BTW I am using 3.3 volt on the Uno for VCC.
The part I bought is:  https://smile.amazon.com/gp/product/B01DLG4NZC/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

I also have a working callback version:

```
from PyMata.pymata import PyMata
import sys
import time
import signal

board = PyMata("/dev/ttyACM0")

# for leonardo
board.i2c_config(0, board.DIGITAL, 3, 2)

# for uno
# board.i2c_config(0, board.ANALOG, 4, 5)

def signal_handler(sig, frame):
    print('You pressed Ctrl+C!!!!')
    if board is not None:
        board.reset()
    sys.exit(0)

signal.signal(signal.SIGINT, signal_handler)

def lux_callback(data):
    datax = data[2]
    print 'Got read data: %s' % data
    lux = (datax[1] << 8 | datax[2]) >> 4
    lux /= 1.2
    print str(lux) + ' lux'

while True:
    try:
        board.i2c_write(0x23, board.I2C_READ_CONTINUOUSLY)  # same results with board.I2C_READ
        time.sleep(.3)
        board.i2c_read(0x23, 0, 2, board.I2C_READ, lux_callback)  # same results with board.I2C_READ_CONTINUOUSLY
        time.sleep(.3)
    except KeyboardInterrupt:
        board.close()
```

MrYsLab, 2017-03-18:
One last note. The I2C_READ_CONTINUOUSLY instruction has to be performed before each read. I tried it once at the beginning and that does not work. Both programs above are working.

Let me know if you have any other questions.

yangmillstheory, 2017-03-18:
Is it a bug in the library that the instruction has to be performed before
each read?

MrYsLab, 2017-03-18:
I based my script off of: https://www.dfrobot.com/wiki/index.php/Bh1750#Sample_Code.

I just added the 2 examples to the distribution and it is on PyPi now.

yangmillstheory, 2017-03-28:
I got the same BH1750 model as you, but using the same script I get:

```
(arduino-playground) arduino-playground git:(master) ✗ python src/bh1750.py

Python Version 2.7.10 (default, Jul 30 2016, 19:40:32)
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.34)]

PyMata version 2.13  Copyright(C) 2013-16 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/cu.usbmodem1421

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 20
Total Number of Analog Pins Detected = 6
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
Got read data: [0]
```

I'm not the world's best solderer, so I may have messed something up when soldering the headers onto the board. I'm powering the BH1750 through 3.3V, have pin 4 connected to SDA and pin 5 connected to SCL, and of course the ground is hooked up as well.

This is a bit discouraging for me; I believe that the code works, I'm just not able to verify it on my end.

MrYsLab, 2017-03-28:
@yangmillstheory
Are you connecting to A4 for SDA and A5 for SCL.? The Uno uses analog pins.

yangmillstheory, 2017-03-28:
Yes, pretty sure those are the connections.

MrYsLab, 2017-03-28:
@yangmillstheory 
I originally tested on a Leonardo. I will retest on an Uno later today and will let you know the results.

MrYsLab, 2017-03-28:
I just tested on an Uno and it works for me. I ran https://github.com/MrYsLab/PyMata/blob/master/examples/i2c/pymata_i2c_read/bh1750_poll.py

I commented out line 8 and uncommented line 11.

yangmillstheory, 2017-03-28:
Can you take a photo of the wiring and paste it here? Thanks for your help!

MrYsLab, 2017-03-28:
Instead of a photo here is a diagram. Pinout:

BH1750 --------------------------Arduino

GND-------------------------------GND
VCC-------------------------------3v3
SDA-------------------------------A4
SCL-------------------------------A5
ADD                        NOT CONNECTED

[image]

MrYsLab, 2017-03-29:
Let me know if this helped when you get a chance to try things out.