MrYsLab/PyMata #32: firmataplus

mingguichen, 2017-04-10:
Thanks for your great work. I got some issues with firmataplus and pymata in Python 2.7. With this setting, I can not start Arduino in Python anymore. DO you have some insights on that? Please see below for the error message.

Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Program Files\Anaconda2\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Program Files\Anaconda2\lib\site-packages\PyMata\pymata_command_handler.py", line 818, in run
    method = dispatch_entry[0]
TypeError: 'NoneType' object has no attribute '__getitem__'

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Program Files\Anaconda2\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 699, in runfile
    execfile(filename, namespace)
  File "C:\Program Files\Anaconda2\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 74, in execfile
    exec(compile(scripttext, filename, 'exec'), glob, loc)
  File "D:/Desktop/Vaziri lab/Francisca/Software/PyTimingTask/PyTimingTask.py", line 57, in <module>
    board.set_sampling_interval(1)
  File "C:\Program Files\Anaconda2\lib\site-packages\PyMata\pymata.py", line 770, in set_sampling_interval
    self._command_handler.send_sysex(self._command_handler.SAMPLING_INTERVAL, data)
  File "C:\Program Files\Anaconda2\lib\site-packages\PyMata\pymata_command_handler.py", line 619, in send_sysex
    self.pymata.transport.write(data)
  File "C:\Program Files\Anaconda2\lib\site-packages\PyMata\pymata_serial.py", line 107, in write
    self.arduino.write(data)
  File "C:\Program Files\Anaconda2\lib\site-packages\serial\serialwin32.py", line 279, in write
    if not self.hComPort: raise portNotOpenError
serial.serialutil.SerialException: Attempting to use a port that is not open
>>>

mingguichen, 2017-04-10:
In my case, I want to use tone as well. That is why I need firmataplus instead of standardfirmata protocol.

MrYsLab, 2017-04-10:
@mingguichen The problem looks like a reset/streaming data issue. If your are able to run your script, but then restart it, the Arduino does not stop sendning data. When your script initializes, PyMata receives unexpected data and an exception is thrown.

Try unplugging the Arduino and plugging it back in, then wait until the lights on the Arduino stop flashing and then start your script.

Please let me know if this solves the issue.

mingguichen, 2017-04-10:
I did everthing based on your suggestions. I still go the sam error message. However, if I run the same program with standardfirmata, everyhting is fine. But I can not use the play_tone().

Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Program Files\Anaconda2\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Program Files\Anaconda2\lib\site-packages\PyMata\pymata_command_handler.py", line 818, in run
    method = dispatch_entry[0]
TypeError: 'NoneType' object has no attribute '__getitem__'

MrYsLab, 2017-04-10:
@mingguichen Could you try running the script in this directory: https://github.com/MrYsLab/PyMata/tree/master/examples/piezo.

Please let me know if that works. You may need to adjust the pin number and com port for your system.

mingguichen, 2017-04-10:
That's actually what I did initially. Still the same message.

PyMata version 2.14  Copyright(C) 2013-17 Alan Yorinks    All rights reserved.

Opening Arduino Serial port COM5 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Program Files\Anaconda2\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Program Files\Anaconda2\lib\site-packages\PyMata\pymata_command_handler.py", line 818, in run
    method = dispatch_entry[0]
TypeError: 'NoneType' object has no attribute '__getitem__'

mingguichen, 2017-04-10:
Thank you for all your effort! I deeply appreciate it!

MrYsLab, 2017-04-10:
@mingguichen Are you running this in anaconda or from a console window? If running from anaconda, please try from a console window.

Also, do you have the latest version of FirmataPlus plus installed? https://github.com/MrYsLab/PyMata/tree/master/ArduinoSketch

Also, which Arduino are you using and which pin are you using?

MrYsLab, 2017-04-10:
@mingguichen If you are still having issues, please try running this simple blink test:
https://github.com/MrYsLab/PyMata/blob/master/examples/pymata_blink.py

mingguichen, 2017-04-10:
Thanks. I pip installed PyMata several days ago. To ensure it's the newest, I just reinstalled everything and uploaded the firmataplus into my Arduino Mega 2560 by modifying the sketchbook location. Then I tried piezo example code you provide again via a digital pin #7. It is still not working.

mingguichen, 2017-04-10:
Worked it out finally. The program you deposited has no issues. My lab mates change my baud rate yesterday to get faster sampling... When I switch it back to 57600, everything works fine. Thank you for all your help. Deeply appreciate it!

MrYsLab, 2017-04-10:
Oh the joys of Firmata ;-).

closed 2017-04-10 by MrYsLab