MrYsLab/PyMata #34: It it possible to change the baud rate?

SapientumGoertz, 2017-10-19:
I need to output a 20hZ sinusoidal-like PWM signal from my Arduino. However I only get around 4hZ and I have a suspicion that I am limited by the baud rate of the connection.

Is it possible to set a different baud rate than the default one using PyMata? I tried looking through the documentation but could not find anything.

MrYsLab, 2017-10-19:
PyMata simply sets the Arduino pin mode to PWM (set_pin_mode) and then sets the user's value (0-255) using analog_write on the Arduino. It is no different than if you wrote a sketch to do the same. 

This article might help you achieve what you need to do: https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM

I am assuming you are trying to create the signal on an Arduino pin and not send a 20 hz signal over the USB serial port.

MrYsLab, 2017-10-19:
I am going to close this issue. If you have any other questions, you may add them to this thread and I will see them.

closed 2017-10-19 by MrYsLab