MrYsLab/PyMata #38: IndexError: list index out of range

Stooney8, 2018-06-22:
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/PyMata/pymata_command_handler.py", line 858, in run
    method(command_data)
  File "/usr/local/lib/python2.7/dist-packages/PyMata/pymata_command_handler.py", line 402, in analog_message
    self.analog_response_table[data[self.RESPONSE_TABLE_MODE]][self.RESPONSE_TABLE_PIN_DATA_VALUE]
IndexError: list index out of range

MrYsLab, 2018-06-22:
Does this occur as a result of restarting your program that previously was receiving analog data?

If yes, it is because you must also physically reset the Arduino before restarting your program. Once an analog pin starts receiving data, data is streamed until the board is reset. When the program is reset without physically resetting the board, PyMata receives data on a pin that was not yet initialized and so can't find an entry in the response table, which causes the list index out of range exception.  Depending upon the model of Arduino, it is not always possible to reset the board through Firmata.

If what I describe is not your problem, please provide some more detail as to what is causing the problem. Also, are you using StandardFirmata

MrYsLab, 2018-06-24:
I am closing this issue since I have not heard back. If you would like to reopen the issue, please
leave a comment and we can continue to discuss the issue.

closed 2018-06-24 by MrYsLab