MrYsLab/PyMata #42: Does FrimataPlus support other client libs?

pkotsiop, 2019-04-12:
Hi,
I just like to know if your version of firmata (FirmataPlus) works with other firmata client libraries like johnny-five. I would like to use PyMata and FirmataPlus on my projects, but my projects also require other languages (javascript mainly) to be able to connect to the same Arduino.

MrYsLab, 2019-04-12:
FirmataPlus for the PyMata library is based on StandardFirmata 2.5.1. It adds some features to that library to support stepper motors, Sonar Ping Devices (HC-SR04), Piezo devices and Rotary Encoders. I don't know if Johnny-Five supports those additional features and if it does, whether or not it is compatible with FirmataPlus. For all other standard Firmata features, it should work. I have not worked with Johnny-Five in several years, so I have no recent data to share.

If you have any specific questions, I would be happy to share any knowledge that I may have.

pkotsiop, 2019-04-12:
I will try it and let you know how it goes. Close the issue and i will post again here when i have more info.

closed 2019-04-12 by MrYsLab

pkotsiop, 2019-04-22:
I have tested FirmataPlus with Johnny-Five (latest nmp of johnny-five for node-red) and the features of standard firmata seem to work just fine.

MrYsLab, 2019-04-22:
Thanks for letting me know.