MrYsLab/PyMata #45: set_pin_mode overwrites previous pin state value

chriski777, 2019-09-17:
Hi,

I ran into this issue where I'm trying to set the pin modes for two pins that correspond to the same pin number but are different types of pins (e.g: digital PWM pin 4 and analog input pin 4).
```
self._board.set_pin_mode(4, PyMata.PWM, PyMata.DIGITAL)
self._board.set_pin_mode(4, PyMata.INPUT, PyMata.ANALOG)
```
However, once I run these two lines, it seems that the second line simply overwrites the first line and I am unable to use the digital PWM pin.

MrYsLab, 2019-09-17:
I can confirm what you are seeing. I will investigate and will provide a fix within the next few days.

MrYsLab, 2019-09-17:
Fixed with release 2.20. Thanks for reporting the issue.

closed 2019-09-17 by MrYsLab

chriski777, 2019-09-17:
works great now. Thanks!